3mdeb/meta-rte #14: tmux fail because of locale settings

pietrushnic, 2021-05-03:
```
root@orange-pi-zero:~# tmux
tmux: need UTF-8 locale (LC_CTYPE) but have ANSI_X3.4-1968
```

I see that on v0.5.3 so it may be duplicate of this: https://github.com/3mdeb/meta-rte/pull/50
but because of some issues I cannot access above pull request - it is mentioned in changelog: https://github.com/3mdeb/meta-rte/blob/master/CHANGELOG.md#v061---2019-01-24

TomaszAIR, 2021-05-11:
tmux is one of the things that has been fixed in version 0.7.0, which will be released after testing is complete. They are scheduled for this week.

The changelog needs to be fixed as links lead to MRs not synced with gitlab.

TomaszAIR, 2021-10-01:
tmux now can be correctly used on the new version (v0.7.1) of the system.

Also links in changelog were fixed, thanks for pointing that out.

The image is available here https://cloud.3mdeb.com/index.php/s/tJy9nZdDG9qd35b, there is no way for update the system from the previous version (like v0.5.3) so the sd card needs to be flashed.

I will close this issue.

closed 2021-10-01 by TomaszAIR