3mdeb/meta-rte #15: Please update flashrom version to

pietrushnic, 2021-05-12:
Please upgrade flashrom based on this request: https://github.com/Dasharo/dasharo-issues/issues/23#issuecomment-838194899

```
root@orange-pi-zero:~# flashrom --version
flashrom v1.0 on Linux 4.18.10 (armv7l)
flashrom is free software, get the source code at https://flashrom.org
```

TomaszAIR, 2021-10-01:
flashrom was upgraded in the newest image for RTE (v0.7.1).

```
root@rte:~# flashrom --version
flashrom v1.2 on Linux 5.4.69 (armv7l)
flashrom is free software, get the source code at https://flashrom.org

root@rte:~# cat /etc/os-release 
ID=rte
NAME="RTE (Remote Test Environment Distro)"
VERSION="0.7.1 (rocko)"
VERSION_ID=0.7.1
PRETTY_NAME="RTE (Remote Test Environment Distro) 0.7.1 (rocko)"
```

The image is available here https://cloud.3mdeb.com/index.php/s/tJy9nZdDG9qd35b, there is no way for update the system from the previous version (like v0.5.3) so the sd card needs to be flashed.

I will close this issue.

closed 2021-10-01 by TomaszAIR

pietrushnic, 2021-10-05:
@TomaszAIR for someone to find where to get new release someone have to find this issue, which is not very helpful. Ideal situation would be if all releases would be published [here](https://docs.dasharo.com/transparent-validation/rte/v1.1.0/reference-os/).

pietrushnic, 2021-10-05:
@TomaszAIR there should be update documentation, in Arch4032 I created some draft for OST2 which could be used for Dasharo documentation: https://gitlab.com/opensecuritytraining/arch4032_coreboot_hw_hands_on_slides_and_subtitles/-/merge_requests/4/diffs#9a1742121855941e81568b4a749bbe5508f2afe0_0_245