3mdeb/meta-rte #27: How to update? Do I have to flash new version to SD?

pietrushnic, 2022-08-14:
**meta-rte version**

0.7.1

**Question text**

Is there a way to update with swupdate or I have to reflash SD card?

pietrushnic, 2022-08-14:
I used copying to SD, but it would be great to have trustworthy update in meta-rte.

What workd on Debian 11:

```shell
% sudo apt install bmap-tools
% sha256sum -c *.sha256
core-image-minimal-orange-pi-zero-v0.7.3.wic.bmap: OK
core-image-minimal-orange-pi-zero-v0.7.3.wic.gz: OK
% sudo bmaptool copy core-image-minimal-orange-pi-zero-v0.7.3.wic.gz /dev/xvdi
bmaptool: info: discovered bmap file 'core-image-minimal-orange-pi-zero-v0.7.3.wic.bmap'
bmaptool: info: block map format version 2.0
bmaptool: info: 266496 blocks of size 4096 (1.0 GiB), mapped 97832 blocks (382.2 MiB or 36.7%)
bmaptool: info: copying image 'core-image-minimal-orange-pi-zero-v0.7.3.wic.gz' to block device '/dev/xvdi' using bmap file 'core-image-minimal-orange-pi-zero-v0.7.3.wic.bmap'
bmaptool: info: 100% copied
bmaptool: info: synchronizing '/dev/xvdi'
bmaptool: info: copying time: 55.7s, copying speed 6.9 MiB/sec
```

pietrushnic, 2022-08-14:
Unfortunately `data` partition was not preserved. This is bug IMHO.

orangecms, 2022-08-16:
On the D1, I found that separating the actual firnware from the distro / OS makes it easier to upgrade the latter, swap it out flexibly, etc..

There is a SPI NOR flash on the OrangePi, so we could put whatever init + U-Boot there, potentially LinuxBoot, and have BLS (boot loader spec) based SD cards (supported by the u-root boot loaders). A shell and/or `cpud` could be a great fallback, i.e., try to boot a distro from SD card, and if none is found, fire up `cpud`.

In the `cpud` case, I can use the RTE arbitrarily without having to install anything on it, upgrading, etc.. I can use my favourite distro via a [Docker image over `cpu`](https://github.com/u-root/cpu#cpu-and-docker), run `RteCtrl` over it, etc.. Then I can also easily make adjustments to `RteCtrl` and need not worry about the OS there.

TomaszAIR, 2023-12-12:
@pietrushnic from some time now we have an update section in README https://github.com/3mdeb/meta-rte#update so maybe we can close it?

BTW, @orangecms in next release we will add `cpud` to the meta-rte

rminnich, 2023-12-12:
neat!

pietrushnic, 2023-12-13:
@TomaszAIR process is complex, but I'm okay with closing this issue. I will request news if I find areas of improvement in update process.

closed 2023-12-13 by pietrushnic