3mdeb/meta-rte #28: Telnet output is empty in random situations

sulewskiprzemyslaw, 2022-08-22:
**meta-rte version**

Every version up from 0.5.3

**Affected component(s) or functionality**

Telnet serial connection

**Brief summary**

An automated test environment is built based on RTE. When using a version of meta-rte newer than 0.5.3, the problems with the Telnet connection appear randomly: the tests fail because of the empty output - even though the DUT is powered on. 

**How reproducible**

Randomly. But if the bug occurs during the test suite preparation, it affects all test cases.

**How to reproduce**

The problem occurs randomly. Under the issue, I attach the logs from automatic tests.

**Expected behavior**

No problems with the opened Telnet connection

**Actual behavior**

Empty output in random situations.

**Screenshots**

**Additional context**

**Solutions you've tried**

1. Improving timeouts between turning on/off the device.

sulewskiprzemyslaw, 2022-08-22:
Under this link, the logs from tests are available: https://cloud.3mdeb.com/index.php/s/ZMFqHj5KaspaCwA.
The bug occurs for example in `Modem2-Detect-Debian` test suite. This test suite is located in `Hardware` test module.

artur-rs, 2024-03-19:
Closing, outdated

closed 2024-03-19 by artur-rs