3mdeb/meta-rte #35: Please include bmaptool in meta-rte

miczyg1, 2023-07-25:
**The problem you're addressing (if any)**
Lack of bmaptool needed by SDWire testing stand

**Describe the solution you'd like**
meta-rte should have bmaptool to flash the SD card for SDWire testing instead of relying on an old Armbian...

**Where is the value to a user, and who might that user be?**
No need to use dd over and over again. Bmaptool will detect quickly if it needs to flash anything on SD. Lesser SD wear out.

TomaszAIR, 2023-07-25:
@miczyg1 thank you for rising an issue. For now we added bmaptool to internal v0.7.4-rc4 https://github.com/3mdeb/meta-rte/tree/v0.7.4-rc4; please let me know if it works as it should.

This OS version also has `sd-mux-ctrl` installed - could you test also that?

PLangowski, 2023-09-22:
@macpijan my PR cherry-picks the changes into master. Is that okay?

PLangowski, 2023-10-03:
Below are logs from the current `develop` branch. 
Running `bmaptool`:

```
# bmaptool -h
usage: bmaptool [-h] [--version] [-q] [-d] {create,copy} ...

A tool for creating block maps (bmap) and copying disk images using bmap files. Documentation can be found here: source.tizen.org/documentation/reference/bmaptool

optional arguments:
  -h, --help     show this help message and exit
  --version      show program's version number and exit
  -q, --quiet    be quiet
  -d, --debug    print debugging information

commands:
  {create,copy}
    create       generate bmap for an image file (which should be a sparse file)
    copy         write an image to a block device using bmap

```
Running `sd-mux-ctrl`:

```
sd-mux-ctrl [-liotdscun?] [-l|--list] [-i|--info] [-o|--show-serial] [-r|--set-serial=STRING] [-t|--init] [-d|--dut] [-s|--ts] [-p|--pins=INT] [-c|--tick] [-u|--status] [-y|--dyper1=STRING] [-z|--dyper2=STRING]
        [-m|--tick-time=INT] [-v|--device-id=INT] [-e|--device-serial=STRING] [-k|--device-type=STRING] [-x|--vendor=INT] [-a|--product=INT] [-n|--invert] [-?|--help] [--usage] command
No command specified
```

PLangowski, 2024-01-05:
We just released [v0.7.4](https://github.com/3mdeb/meta-rte/releases/tag/v0.7.4), where we introduce bmaptool. We can close the issue.

closed 2024-01-16 by artur-rs