3mdeb/meta-rte #36: Script for resizing data partition

TomaszAIR, 2023-07-26:
**The problem you're addressing (if any)**

data partition is set to 512M and most of the time system boots from SD cards that have 16G/32G which left us with at least ~12G of not used space that could be used e.g. as images storage

**Describe the solution you'd like**

a script that on boot would check if there is a possibility to resize last partition and if there is, does it

**Where is the value to a user, and who might that user be?**

lot of space for data

**Describe alternatives you've considered**

preparing new partition layout, but this script could be provided by OTA and works on machines without reflashing whole image

**Additional context**

pietrushnic, 2023-07-26:
This issue and recent interest in RTE brings the idea that RTE could become self-hosted all-in-one open-source firmware CI/CD, validation, and deployment system, which is related to my vision expressed in https://github.com/Dasharo/dasharo-issues/issues/468#issuecomment-1643696857

pietrushnic, 2023-07-26:
We could consider NLNet application for this @macpijan