3mdeb/meta-rte #38: We are using old config file format for ser2net

macpijan, 2023-08-07:
**meta-rte version**

v.0.7.1

**Affected component(s) or functionality**

ser2net

**Brief summary**

ser2net switched to yaml config since v4.0.0 version

**How reproducible**

100%

**How to reproduce**

Steps to reproduce the behavior:
1.  journalctl -fu ser2net

**Expected behavior**

No errros / warnings regarding the config are displayed

**Actual behavior**

May 26 11:56:29 rte ser2net[179]: ser2net:WARNING: Using old config file format, this will go away
May 26 11:56:29 rte ser2net[179]: soon.  Please switch to the yaml-based format.

**Screenshots**

N/A

**Additional context**

N/A
**Solutions you've tried**

johanes2115, 2023-09-22:
@PLangowski I added a new connection to the default config file in order to read from ttyACM0:
`# Connection to read from /dev/ttyACM0
connection:
accepter: telnet, tcp, 2002
connector: serialdev, /dev/ttyACM0, 9600n81, local`

[ser2net.zip](https://github.com/3mdeb/meta-rte/files/12698578/ser2net.zip)

macpijan, 2023-09-22:
If you want to test, you should connect something to the given serial port device and on your PC use telnet command with RTE IP and reelvant port, e.g.

```
telnet 192.168.10.228 13541 
```

If you can access this serial over IP, it works.

macpijan, 2023-09-22:
Unfortunately the config attached above is plain wrong and results in parsing error when ser2net get started, so it was not tested.

The proper relevant section of the /etc/ser2net/ser2net.yaml` looks like follows:

```
# Create a connection to read from /dev/ttyACM0
connection: &con00
  accepter: tcp,13541
  connector: serialdev,/dev/ttyACM0,115200n81,local
```

PLangowski, 2023-09-27:
@macpijan I don't quite understand. My sections look like that except I changed the serial device names so that they match the previous config and I add a timeout, which I believe was the meaning behind `600` in the `.conf` file. Am I missing something?

PLangowski, 2023-09-28:
@macpijan the current configuration in the PR above has been tested and works.

PLangowski, 2023-09-29:
I think the problem in [this comment](https://github.com/3mdeb/meta-rte/issues/38#issuecomment-1731337190) was caused by faulty RTE. That one did not even work with the old config - there was no output.

I changed the RTE to a different one. I connected it via USB-UART to another platform. I flashed it with the image from my branch, then from within Linux I changed `/dev/ttyS1` to `/dev/ttyUSB0` in `ser2net.yaml`, so that I would get output from USB-UART. The parse errors did not appear:
```
# systemctl status ser2net
● ser2net.service - ser2net
     Loaded: loaded (/lib/systemd/system/ser2net.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-09-27 11:42:43 UTC; 1 day 23h ago
    Process: 196 ExecStart=/usr/sbin/ser2net -c /etc/ser2net.yaml -P /run/ser2net.pid (code=exited, status=0/SUCCESS)
   Main PID: 206 (ser2net)
     CGroup: /system.slice/ser2net.service
             └─206 /usr/sbin/ser2net -c /etc/ser2net.yaml -P /run/ser2net.pid

Sep 27 11:42:43 rte systemd[1]: Starting ser2net...
Sep 27 11:42:43 rte ser2net[196]: Unable to start mdns: Operation not supported
Sep 27 11:42:43 rte systemd[1]: Started ser2net.
```

Then I was able to get output from the other platform by connecting to the tcp port from my host PC.

```
$ telnet 192.168.4.134 13541
Trying 192.168.4.134...
Connected to 192.168.4.134.
Escape character is '^]'.

U-Boot SPL 2023.01-g62e2ad1 (Jan 09 2023 - 16:07:33 +0000)
DRAM: 1024 MiB
CPU: 912000000Hz, AXI/AHB/APB: 3/2/2
Trying to boot from MMC1

U-Boot 2023.01-g62e2ad1 (Jan 09 2023 - 16:07:33 +0000) Allwinner Technology

DRAM:  1 GiB
Core:  51 devices, 19 uclasses, devicetree: separate
WDT:   Not starting watchdog@1c20c90
MMC:   mmc@1c0f000: 0
Loading Environment from FAT... OK
Unknown monitor
Unknown monitor
In:    serial
Out:   serial
Err:   serial
Unknown command 'usb' - try 'help'
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1439 bytes read in 1 ms (1.4 MiB/s)
## Executing script at 43100000
Saving Environment to FAT... OK
Saving Environment to FAT... OK
27044 bytes read in 3 ms (8.6 MiB/s)
11170224 bytes read in 464 ms (23 MiB/s)
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-6.1.9
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    11170160 Bytes = 10.7 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
Working FDT set to 43000000
   Loading Kernel Image
   Loading Device Tree to 49ff6000, end 49fff9a3 ... OK
Working FDT set to 49ff6000
DE is present but not probed

Starting kernel ...

(...)
```

So in my opinion, that was an issue with a bad RTE, not with my config. I could check more RTEs if you think that is necessary.

TomaszAIR, 2023-10-03:
> I could check more RTEs if you think that is necessary.

@PLangowski there is no need for that; I merged changes

PLangowski, 2024-01-05:
The config has been updated in [v0.7.4](https://github.com/3mdeb/meta-rte/releases/tag/v0.7.4). Can we close this issue?

closed 2024-01-16 by artur-rs