3mdeb/meta-rte #51: RTE dmesg show warning related to "buggy DT: spidev listed directly in DT"

bdelgado1995, 2023-12-15:
**meta-rte version**

v1.1.0

**Question text**
My two RTEs show the warning below in the system dmesg for the spidev. Is this harmless or a sign of a problem? It looks like the spidev0.0 detects a chip but I haven't been able to get the spidev1.0 able to detect the chip, but that could also be for some other issue. But for now, just wanted to see if this warning indicates a problem or if it's harmless. 

``` 0.942965] WARNING: CPU: 0 PID: 1 at drivers/spi/spidev.c:737 spidev_probe+0xf0/0x1ac```
``` [    0.950930] /soc/spi@1c68000/spidev@0: buggy DT: spidev listed directly in DT ```

The log is attached. 
[spidev-rte.txt](https://github.com/3mdeb/meta-rte/files/13690122/spidev-rte.txt)

artur-rs, 2024-01-16:
@bdelgado1995 thank you for raising this issue. We've scheduled the analysis for this week. The result should be added to the FAQ/Troubleshooting section in the README/Dasharo documentation. I'll keep you updated.

PLangowski, 2024-01-19:
@bdelgado1995 Hi, the warning was added to `spidev.c` to deter people from setting the `compatible` attributes in the device tree to `spidev` ([source](https://patchwork.kernel.org/project/spi-devel-general/patch/1427499742-26916-1-git-send-email-broonie@kernel.org/)). The warning is harmless. We actually tried to work around it before, but there was a slight issue with the way we did it. We have fixed the problem in https://github.com/3mdeb/meta-rte/pull/55. It will soon be merged into develop. Thank you for raising the issue.

PLangowski, 2024-03-13:
@bdelgado1995 Hi again, just letting you know that the problem is now fixed in [release v0.7.5](https://github.com/3mdeb/meta-rte/releases/tag/v0.7.5)

closed 2024-03-13 by artur-rs

bdelgado1995, 2024-03-13:
That's great to hear! Thanks for the closure.