3mdeb/meta-rte #58: ser2net service should be improved to watch for 13541 port

pietrushnic, 2024-03-04:
**meta-rte version**

0.7.4

**Affected component(s) or functionality**

ser2net

**Brief summary**

Currently for some reason port 13541 may not exposed even if service is running without issues:

```bash
root@rte:~# systemctl status ser2net
WARNING: terminal is not fully functional
● ser2net.service - ser2net
     Loaded: loaded (/lib/systemd/system/ser2net.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2024-02-28 09:48:14 UTC; 5 days ago
    Process: 180 ExecStart=/usr/sbin/ser2net -c /etc/ser2net.yaml -P /run/ser2net.pid (code=exited, status=0/SUCCESS)
   Main PID: 186 (ser2net)
     CGroup: /system.slice/ser2net.service
             └─186 /usr/sbin/ser2net -c /etc/ser2net.yaml -P /run/ser2net.pid

Feb 28 09:48:14 rte systemd[1]: Starting ser2net...
Feb 28 09:48:14 rte ser2net[180]: Unable to start mdns: Operation not supported
Feb 28 09:48:14 rte ser2net[180]: Invalid port name/number: Invalid data to parameter on line 11 column 0
Feb 28 09:48:14 rte ser2net[180]: Invalid port name/number: Invalid data to parameter on line 14 column 0
Feb 28 09:48:14 rte systemd[1]: Started ser2net.
root@rte:~# cat /etc/ser2net.yaml
%YAML 1.1
---

define: &confver 1.0

# Connection to read from /dev/ttyS1
connection: &con1
  accepter: telnet, tcp, 13541
  connector: serialdev, /dev/ttyS1, 115200n81, local

# Connection to read from /dev/debug_uart_converter
connection: &con2
  accepter: telnet, tcp, 13542
  connector: serialdev, /dev/debug_uart_converter, 115200n81, local
root@rte:~# netstat -an |grep 13541
root@rte:~#
```

It requires restart to make things work, what should happen automatically.

**How reproducible**

Sometimes

**How to reproduce**

Steps to reproduce the behavior:
1. ssh to platform
2. check if 13541 is exposed in netstat output
3. restart service and check again

**Expected behavior**

**Actual behavior**

**Screenshots**

**Additional context**

There are also other minor issues with the service (thanks to @miczyg1 ):
- /dev/debug_uart_converter will never be a valid devnode
- Current config will be different per platform. Some will have /dev/ttyS1 as primary connection and /dev/ttyUSB0 as secondary (or empty), OR /dev/ttyUSB0 as primary connection and secondary empty

**Solutions you've tried**

macpijan, 2024-03-04:
> /dev/debug_uart_converter will never be a valid devnode

It will be if you create a proper udev rule to create it.

It is especially useful (even required) where you have more than 1 ttyUSB devices to have persistent naming.

macpijan, 2024-03-04:
> Currently for some reason port 13541 may not exposed even if service is running without issues:

I think I saw such behavior when nothing was connected. Eg. ttyUSB0 configured, ser2net does not complain even if the device is not there, but you cannot connect.

pietrushnic, 2024-03-04:
> It will be if you create a proper udev rule to create it.

I guess that confirms there is a problem with meta-rte because it should deliver such rules by default.

> I think I saw such behavior when nothing was connected. Eg. ttyUSB0 configured, ser2net does not complain even if the device is not there, but you cannot connect.

Yes, but in my case restarting service fixed the issue, so apparently serial is connected.

macpijan, 2024-03-04:
> I guess that confirms there is a problem with meta-rte because it should deliver such rules by default.

It does ship the above example (2 entries, the other two have been added for this particular case) currently. There is no one universal rule here. Depends on the platform. And even more, on specific adapters used on the test stand.
I think at best right now it can be documented, at least.

PLangowski, 2024-04-02:
I have added information about this to the documentation https://github.com/Dasharo/docs/pull/798

closed 2024-04-02 by macpijan