3mdeb/meta-rte #64: Integrate Dediprog SF100 flashing software

miczyg1, 2024-06-27:
**The problem you're addressing (if any)**
Some platforms do not flash well with flashrom programmer linux_spi. Somtimes even flashrom dediprog programmer has problems with properly flashing the device. However, the dedicated Linxu software for Dediprog programmers successfully updates the flash contents compared to flashrom.

**Describe the solution you'd like**
Integrate the https://github.com/DediProgSW/SF100Linux software into RTE distro for cases where dpcmd can provide reliable flashing method, alternative to flashrom.

**Where is the value to a user, and who might that user be?**
All 3mdeb employees that need to work on hardware that is very tricky to setup remotely and only a specific programmers work with it.

**Describe alternatives you've considered**
Are there any alternatives? The flashing always fails with some single byte/bit errors no matter what, but dpcmd works well...

**Additional context**
None

macpijan, 2024-06-28:
The recipe should be upstreamed, and in meta-rte we should simply use it.

tym2k1, 2024-11-27:
The [`dediprog` utility was added](https://github.com/3mdeb/meta-rte/tree/master/recipes-devtools/dediprog-flasher) to the image with the `v0.8.0` release as well as accepted upstream. I've filed an [issue for deleting the recipe from our layer and using the one in the `meta-openembedded`](https://github.com/3mdeb/meta-rte/issues/85). Closing this one as it's done

closed 2024-11-27 by tym2k1