3mdeb/meta-rte #72: `openvpn` implementation doesn't seem right

tym2k1, 2024-07-16:
**meta-rte version (if applicable)**

`v0.7.5` and onward

**Affected component(s) or functionality (if applicable)**

OpenVPN

**Brief summary**

## 1. Misaligned references

The [openvpn_%.bbappend](https://github.com/3mdeb/meta-rte/blob/master/recipes-support/openvpn/openvpn_%25.bbappend#L5) recipe specifies that the config for the rte openvpn service should be at `/data/openvpn/`
 while the `openvpn@.service` we implement [checks for `/storage/openvpn/rte.conf`](https://github.com/3mdeb/meta-rte/blob/master/recipes-support/openvpn/openvpn_%25.bbappend#L5) which doesn't exist.
 
 [We also use `sed` to replace `openvpn` references to `/etc` with `/storage`](https://github.com/3mdeb/meta-rte/blob/master/recipes-support/openvpn/openvpn_%25.bbappend#L9-L11) while README.md specifies [putting the rte.conf on the `/data` partition](https://github.com/3mdeb/meta-rte?tab=readme-ov-file#openvpn).
 
## 2. The service seems broken on version `0.7.5`

`cat /ets/os-release`
```
ID=rte
NAME="RTE (Remote Test Environment Distro)"
VERSION="0.7.5 (dunfell)"
VERSION_ID=0.7.5
PRETTY_NAME="RTE (Remote Test Environment Distro) 0.7.5 (dunfell)"
```
`systemctl status openvpn@rte`
```
● openvpn@rte.service - Our version of openvpn service
     Loaded: loaded (/lib/systemd/system/openvpn@.service; disabled; vendor preset: disabled)
     Active: inactive (dead)
                                                                                                                                                                                                                   
Jul 16 14:27:00 rte systemd[1]: /lib/systemd/system/openvpn@.service:9: PIDFile= references a path below legacy directory /var/run/, updating /var/run/openvpn/rte.pid → /run/openvpn/rte.pid; please update the unit file accordingly.
```

When starting the service with `systemctl start openvpn@rte`:
```
● openvpn@rte.service - Our version of openvpn service
     Loaded: loaded (/lib/systemd/system/openvpn@.service; disabled; vendor preset: disabled)
     Active: inactive (dead)

Jul 16 14:35:35 rte systemd[1]: /lib/systemd/system/openvpn@.service:9: PIDFile= references a path below legacy directory /var/run/, updating /var/run/openvpn/rte.pid → /run/openvpn/rte.pid; please update the unit file accordingly.
Jul 16 14:35:35 rte systemd[1]: Condition check resulted in Our version of openvpn service being skipped.
Jul 16 14:35:35 rte systemd[1]: /lib/systemd/system/openvpn@.service:9: PIDFile= references a path below legacy directory /var/run/, updating /var/run/openvpn/rte.pid → /run/openvpn/rte.pid; please update the unit file accordingly.
Jul 16 14:35:36 rte systemd[1]: /lib/systemd/system/openvpn@.service:9: PIDFile= references a path below legacy directory /var/run/, updating /var/run/openvpn/rte.pid → /run/openvpn/rte.pid; please update the unit file accordingly.
```
 
 ## 3. `scarthgap`
 
 When working on https://github.com/3mdeb/meta-rte/pull/70 I keep getting an error:
 
 ```bitbake
ERROR: openvpn-2.6.10-r0 do_package: Didn't find service unit 'openvpn@rte.service', specified in SYSTEMD_SERVICE:openvpn-rte. Also looked for service unit 'openvpn@.service'.
ERROR: Logfile of failure stored in: /build/tmp/work/cortexa7t2hf-neon-rte-linux-gnueabi/openvpn/2.6.10/temp/log.do_package.39524
ERROR: Task (/build/../work/meta-openembedded/meta-networking/recipes-support/openvpn/openvpn_2.6.10.bb:do_package) failed with exit code '1'
```
 
 It seems we must implement some default `rte.conf` and ensure that it's disabled by default

macpijan, 2024-07-16:
Let's just drop it if it causes too much trouble. I thought I suggested elsewhere that we do not use openvpnv, so we can remove drop it, or simply keep installed in image, but drop all customizations.

tym2k1, 2024-07-16:
Then i'll remove all customization so it uses the standard recipe from poky and ensure that the service is disabled by default. If someone needs they'll be able to add their config and start the service.

macpijan, 2024-07-16:
Makes perfect sense to me.

artur-rs, 2024-12-03:
@tym2k1 manual check needed - please confirm/resolve the issue.

artur-rs, 2025-01-31:
@EduKav1813 ^