3mdeb/meta-rte #79: Integrate DCU support

wessel-novacustom, 2024-10-31:
**The problem you're addressing (if any)**

Currently, it is not possible to modify firmware binaries.

**Describe the solution you'd like**

Being able to modify firmware binaries with the DCU tool from the RTE programmer directly, for example injecting a serial number.

**Where is the value to a user, and who might that user be?**

1. We can optimise the automatisation of our production process.
2. The correct firmware binary can be applied with an own script that determines the binary from the serial number.
3. We could check the warranty status of a customer's laptop more easily.

macpijan, 2024-11-05:
@artur-rs  Let's take a look on how to integrate the `dcu` script from: https://github.com/Dasharo/dcu into RTE OS.

We already have a recipe for it, as it is included in meta-dts: https://github.com/Dasharo/meta-dts/blob/main/meta-dts-distro/recipes-dasharo/dasharo-configuration-utility/dasharo-configuration-utility_git.bb

We should consider moving `dcu` recipe, and perhaps some dependencies as well, to another layer(s) that can be reused across both DTS and RTE projects.

macpijan, 2024-11-07:
We have created the following layers:
- https://github.com/Dasharo/meta-dasharo
- https://github.com/zarhus/meta-coreboot

We have created a spreadsheet to help us to drive the layers' decomposition: https://pad.3mdeb.com/sheet/#/2/sheet/view/roMqhd-jY+4-dM-98AAujCwEXfj-sVQ6FCoWDy8UM+8/

Execution steps:
- [ ] Initialize `meta-dasharo` and `meta-coreboot` as Yocto layers (https://pad.3mdeb.com/sheet/#/2/sheet/view/roMqhd-jY+4-dM-98AAujCwEXfj-sVQ6FCoWDy8UM+8/)
- [ ] Move recipes from `meta-dts-distro` to `meta-coreboot` and `meta-dasharo` according to spreadsheet
- [ ] Add `meta-coreboot` and `meta-zarhus` to `meta-rte` build
     - drop `coreboot-utils` recipe from `meta-rte` - use recipes from `meta-coreboot` instead
     - install `dcu` package to the RTE image
- [ ] Add `meta-coreboot` and `meta-zarhus` to `meta-dts-distro` build
     - no further changes in the layer should be required, we simply moved some of the recipes to the external layers
- [ ] Add both new layers to index: https://layers.openembedded.org/layerindex/branch/master/layers/
- [ ] Review coreboot documentation to see if there is a proper section where we could mention this `meta-coreboot` layer: https://doc.coreboot.org/community/index.html#

Al-an-21, 2024-11-14:
@macpijan I ran `pre-commit run --all-files` to check if everything passed, and it flagged several issues, like spelling errors and indentation warnings. Do I need to fix all of these manually, or is there a more efficient way to handle these corrections?

Al-an-21, 2024-11-14:
> @macpijan I ran `pre-commit run --all-files` to check if everything passed, and it flagged several issues, like spelling errors and indentation warnings. Do I need to fix all of these manually, or is there a more efficient way to handle these corrections?

`aostrowski in ~/meta-dasharo on move-recipes ● ● ● λ pre-commit run --all-files
check for added large files..............................................Passed
check for merge conflicts................................................Passed
check for broken symlinks............................(no files to check)Skipped
detect private key.......................................................Passed
fix end of files.........................................................Passed
trim trailing whitespace.................................................Passed
mixed line ending........................................................Passed
codespell................................................................Failed
- hook id: codespell
- exit code: 65

dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:19: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:20: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:21: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:22: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:23: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:24: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:25: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:26: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:27: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:28: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:29: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:30: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:31: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:32: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:33: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:34: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:35: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:36: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:37: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:38: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:39: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb.bak:40: crate ==> create
fwupd/fwupd_2.0.1.bb.bak:30: hsi ==> his
fwupd/fwupd_2.0.1.bb.bak:65: hsi ==> his
txe-secure-boot/files/a5df001.diff:6: ths ==> the, this
txe-secure-boot/files/a5df001.diff:16: understant ==> understand, understate
fwupd/fwupd_2.0.1.bb:30: hsi ==> his
fwupd/fwupd_2.0.1.bb:65: hsi ==> his
dasharo-ectool/dasharo-ectool_0.3.8.bb:19: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:20: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:21: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:22: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:23: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:24: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:25: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:26: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:27: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:28: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:29: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:30: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:31: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:32: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:33: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:34: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:35: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:36: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:37: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:38: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:39: crate ==> create
dasharo-ectool/dasharo-ectool_0.3.8.bb:40: crate ==> create

yamllint.............................................(no files to check)Skipped
Name spellchecker........................................................Passed
Advanced oelint..........................................................Failed
- hook id: oelint-adv
- exit code: 1
- files were modified by this hook

/home/aostrowski/meta-dasharo/dasharo-configuration-utility/dasharo-configuration-utility_git.bb:10:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
/home/aostrowski/meta-dasharo/dasharo-configuration-utility/dasharo-configuration-utility_git.bb:21:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
/home/aostrowski/meta-dasharo/dasharo-ectool/dasharo-ectool_0.3.8.bb:9:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
/home/aostrowski/meta-dasharo/dasharo-ectool/dasharo-ectool_0.3.8.bb:40:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
/home/aostrowski/meta-dasharo/txe-secure-boot/smmstoretool_git.bb:6:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
[... cut ...]
/home/aostrowski/meta-dasharo/txe-secure-boot/txesbmantool_git.bb:6:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
/home/aostrowski/meta-dasharo/txe-secure-boot/txesbmantool_git.bb:12:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
/home/aostrowski/meta-dasharo/txe-secure-boot/txesbmantool_git.bb:41:warning:oelint.vars.multilineident:On a multiline assignment, line indent is desirable. 2 set, 4 desirable [branch:true]
`

macpijan, 2024-11-14:
Many of them should be autofixed already after you run this command.

Others should be quick to fix via editor global replace feature, or sed.

In general, one can also call `codespell` command locally in interactive mode, to fix some of the typos.

The fwupd is likely false-positive nad must be excluded, e.g. via codespellx.

macpijan, 2024-12-13:
The DCU integration in meta-rte is finalized via these Pull Requests:
- https://github.com/Dasharo/meta-dts/pull/192
- https://github.com/3mdeb/meta-rte/pull/84
- https://github.com/zarhus/meta-coreboot/pull/2
- https://github.com/Dasharo/meta-dasharo/pull/4

This will be ready to be included in the next release of the RTE OS image.

mkopec, 2024-12-20:
@wessel-novacustom 

The feature on firmware side is implemented here: https://github.com/Dasharo/coreboot/pull/605

However please be aware that the current implementation in DCU injects the S/N and UUID into firmware in such a way that it invalidates Vboot signatures and requires re-signing. This means that every user will have to migrate smbios data and re-sign binaries on their devices. It can be done automatically in DTS but it means everyone will be using selfsigned binaries instead of Dasharo-signed binaries. This is not optimal for security.

Instead, this feature could be implemented in such a way that S/N and UUID are stored in a separate, unsigned region of the flash, exactly like persistent bootsplash is solved. This would avoid this problem altogether. DTS and capsule update code would have to be modified to perform UUID and SN migration but that would be safer and simpler than re-signing binaries.

Another solution would be to abandon Vboot and use CBFS verification, but that would remove Dasharo signatures from the firmware. This is is a bigger topic that is discussed here: https://github.com/Dasharo/dasharo-issues/issues/1074

wessel-novacustom, 2024-12-21:
All right, I am now awaiting those improvements before using this feature.

wessel-novacustom, 2025-02-18:
BIOS logo changing is now possible with DCU from DTS directly.

We would need S/N integration, but that's part of DCU. DCU support has been integrated in DTS, which is why I'm closing this ticket.

closed 2025-02-18 by wessel-novacustom