3mdeb/meta-rte #82: Design of replacing flash.sh script with Python script reusing OSFV lib

macpijan, 2024-11-05:
We do already have some Python module: https://github.com/Dasharo/osfv-scripts/blob/main/osfv_cli/src/osfv/libs/rte.py
which already implements what we need in this script: https://github.com/3mdeb/RteCtrl/pull/22/files

We should review both of them, and propose if/how we can achieve code reuse, by implementing local `flash.py` script on RTE, using the same module, simply using localhost address as `rte_ip`.

This script may reside in `osfv-scripts` repo.

macpijan, 2024-11-05:
Maybe there is no need for another script, and we could somehow use `osfv_cli` directly on the localhost.

I think resolving this https://github.com/Dasharo/osfv-scripts/issues/57 and possible also adding one or two more features, could help to achieve this.

We need to first take a look at possible solutions and decide on something.

DaniilKl, 2024-11-25:
It could be done by installing `osfv_cli` on RTE and using `localhost` as RTE IP. But `osfv_cli` as well as `RteCtrl` will need to be extended for WP features and for using custom `flashrom` parameters in the way `flash.sh` does. Currently it is not implemented. We can also consider using SSH or REST API to do the work, though IMO REST API is more suitable to use with `localhost`≤ but in that case, `osfv_cli` will need to be modified to use `flashrom` via REST API, because currently it is using SSH.

And, most importantly, the issue https://github.com/Dasharo/osfv-scripts/issues/57, must be solved, otherwise this issue could not be solved.

macpijan, 2024-12-02:
> Make osfv_cli use flashrom via HTTP instead of SSH for localhost (optional).

Could be done, but what would be the main benefit? I would also avoid switching to HTTP **just** for localhost. It would need to be done in general. BTW, we already have some of it implemented: https://github.com/3mdeb/RteCtrl/tree/master/pkg/flashromControl

> Add providing additional parameters to flashrom

Correct, at minimum some additional flag to `flash` commmand would do.

>  Add WP options according to flash.sh script

This can in fact be added, but no currently active platforms needs it AFAIK. We do not use this option of this script. Low priority.

DaniilKl, 2024-12-20:
> Could be done, but what would be the main benefit?

I do not see any disadvantages of the SSH solution, therefore marked it as `optional`. We could skip it if you think it is unnecessary.

WiktorG351, 2025-01-23:
@macpijan does this approach seem right?
https://github.com/Dasharo/osfv-scripts/pull/97

DaniilKl, 2025-01-27:
Bruh, my bad. Replace `0.0.0.0` with `127.0.0.1`.

WiktorG351, 2025-02-05:
@DaniilKl thanks for the heads up. So to do this we will need to have `osfv_cli` on the RTE - I have started work on a recipe for `osfv_cli` in meta-rte in [this PR](https://github.com/3mdeb/meta-rte/pull/98) - although its not working just yet.