3mdeb/meta-rte #85: Use upstreamed version of dediprog utility

tym2k1, 2024-11-27:
**meta-rte version (if applicable)**

`v0.8.0`

**Affected component(s) or functionality (if applicable)**

`dediprog` flashing tool

**Brief summary**

We've added the [DediProgSW/SF100Linux](https://github.com/DediProgSW/SF100Linux) as a bitbake recipe in release `v.0.8.0`. The recipe has also been [accepted into `meta-openembedded`](https://layers.openembedded.org/layerindex/recipe/414626/) and the [required patches](https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-bsp/dediprog-flasher/dediprog-flasher) were accepted in the SF100Linux repo itself [[1]](https://github.com/DediProgSW/SF100Linux/pull/86) [[2]](https://github.com/DediProgSW/SF100Linux/pull/88) and included in the `1.14.21` release. The best course of action would be to bump the version in `meta-openembedded` and use the dediprog version from there in our layer.

m-iwanicki, 2024-12-10:
Master branch is compatible with `styhead` and `walnascar` so we would need to update Yocto version again, probably to `styhead` as `meta-sunxi` isn't compatible with `walnascar` from what I can see.

tym2k1, 2025-01-28:
I've originally added this issue only for tracking purposes. I don't think it should really be touched untill we're updating yocto/poky versions to the next LTS.